ankitdhall/lidar_camera_calibration #128: The issue of the launch file

yulan0215, 2020-08-09:
I used the launch file but when I started the launch file there was a mistakes like this:
[aruco_mapping-1] process has died [pid 4229, exit code -6, cmd /home/test/catkin_ws/devel/lib/aruco_mapping/aruco_mapping /image_raw:=/camera/image_raw __name:=aruco_mapping __log:=/home/test/.ros/log/25692aae-da22-11ea-8a07-180373c74b08/aruco_mapping-1.log].
log file: /home/test/.ros/log/25692aae-da22-11ea-8a07-180373c74b08/aruco_mapping-1*.log

Could you please tell me what is the proble? Besides, I used the avt-camera(mono camere) and lidar is Ouster lidar, is it feasible or not? Thank you very much!

Andyyyyyyyyyyyy, 2020-08-14:
May be you should check the .ini file in /aruco_mapping/data/

Gao-guang, 2020-09-01:
the same issue with me ,have you solve it?

yulan0215, 2020-09-01:
> the same issue with me ,have you solve it?

Not yet...

dkhanna511, 2020-11-05:
I have the same issue. I'm not able to get what to change in aruco_mapping.launch. I want to use Point grey camera.