ankitdhall/lidar_camera_calibration #130: Error while calibrating camera-lidar

poornimajd, 2020-09-30:
Great work @ankitdhall and team!
I tried using this implementation with velodyne-32C and Zed camera.This is my setup-
[image]
I have changed the topic names in the config file ,the marker coordinates ,and uncommented aruco-mapping in launch file.
I am using live sensor feed from lidar and camera,and when I played in rviz,it displays both the image and point cloud properly.(shown below)
[image]
This indicates that there is time sync between camera and lidar.
But eachtime I run the command -
`roslaunch lidar_camera_calibration find_transform.launch`

I get the following error
[image]
Only Mono8 window displays,the lidar (pointcloud window) is not displayed.
I checked the topics getting published, the lidar_camera_calibration_rt is getting published but when I echo it gives the output as - are your messages being built?
My question is when ever I run the command why does the process die and throw that error as shown in the picture.
I checked out similar issues but I was unable to get a solution.
Any suggestion is greatly appreciated.

yulan0215, 2020-10-05:
I have the same problem, do you solve it now? Thanks

poornimajd, 2020-10-05:
No,not yet.

dkhanna511, 2020-11-04:
I have the same problem. I built the packages required. And launched the point grey camera and VLP 16 Lidar. However, I haven't really created the setup for now and I ran the find_transform.launch file and I'm having this issue. Can you let me know what result is expected if we connect everything without the setup and launch the file?

miriamrebekah, 2021-04-09:
@ankitdhall, I have encountered the same issue. "point_cloud" seems to go out of scope inside instensityByRangeDiff(point_cloud,config).  By pasting the intensityByRangeDiff(point_cloud,config) function body into your code, the error gives more detail. After looping several times through the pointcloud, the function throw a std::bad_alloc.

ankitdhall, 2021-04-12:
@miriamrebekah thanks for the detailed log. I am not actively maintaining this repo; could you create a merge request that solves this issue that you encounter?

cppchuff, 2022-05-05:
I met the issue when I use the rslidar-32, do you solve it? Many thanks.

cppchuff, 2022-05-12:
https://github.com/ankitdhall/lidar_camera_calibration/blob/13d52954fa18ee3eef86272757555a28a2532c71/include/lidar_camera_calibration/PreprocessUtils.h#L200.
change the num from 16 to 32. It will be work.