ankitdhall/lidar_camera_calibration #133: Every time I mark the four corners of the first side, it says "[find_transform-2] process has died"

JianzhongLEE, 2021-05-10:
Hello, why do I report an error every time I mark the four corners of the first side
[image]

JianzhongLEE, 2021-05-10:
This error has nothing to do with Chinese characters. It was reported before I added some cout

JianzhongLEE, 2021-05-10:
Here is the picture captured just before it reported errors
[image]

YinghuiYou, 2021-09-25:
兄弟，你怎么生成的ArUco标签，我生成的都检测不到

rrkini, 2022-02-23:
Hi @JianzhongLEE  were you able to find the cause for this error?

stanny880913, 2024-02-22:
Hi, may I ask how to click the corners? I press left key and space , these is nothing happen in polygon!