ankitdhall/lidar_camera_calibration #135: Do you provide the Bag?

yanlong658, 2021-07-05:
Hello sir,

I am interested in your work and want to try your code.
Do you provide the Bag?

Wish you a happy week.

yanlong658, 2021-07-13:
> 先生，你好，
> 
> 我對你的工作很感興趣，想試試你的代碼。
> 你們提供袋子嗎？
> 
> 祝你一周愉快。

yanlong658, 2021-07-14:
> Hello sir,
> 
> I am interested in your work and want to try your code.
> I don't know why I can't calibrate it.
> Would you help me to check?
> 
> The following is my video.
> https://www.youtube.com/watch?v=h8L2jrAjZmA
> 
> Wish you a happy week.

louislelay, 2024-07-26:
Hi @yanlong658, in your video, it doesn't look like you've understood the correct way to click. For each line, you should click 4 times to create a quadrilateral that will enclose the points of a segment. And remember that after each click, you should press a key (like `a` for example). Here's an image of what a quadrilateral should look like. 

[image]

I hope this helps you!