ankitdhall/lidar_camera_calibration #142: TF to MSG: Quaternion Not Properly Normalized ??

Wuxinxiaoshifu, 2022-03-22:
[ERROR] [1647922884.020933792]: Not able to lookup transform
[ WARN] [1647922884.020993380]: TF to MSG: Quaternion Not Properly Normalized
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "camera_0" from authority "unknown_publisher" because of a nan value in the transform (-nan -nan -nan) (nan nan nan nan)
         at line 244 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_core.cpp
Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "camera_0" from authority "unknown_publisher" because of an invalid quaternion in the transform (nan nan nan nan)

algorithmsummer, 2022-04-27:
Hi, do you have solved the problems?

0NNS0, 2022-12-06:
我也遇到了这个问题，然后得到了解决。
我遇到这个问题的情况是：我在电脑上对录制的rosbag进行标定。
解决问题的情况是：我在电脑上对传感器进行实时的标定。
原因我认为是：如果对录制的rosbag进行标定，雷达的时间戳是录制时的时间戳，而aruco mapping的时间戳是当前运行程序时的时间戳，所以TF不会进行转化。
所以为了避免此问题的出现，必须对传感器进行实时的标定，而非录制bag后再进行标定。