ankitdhall/lidar_camera_calibration #148: launching aruco_mapping resulted in this error message?

zat1999, 2023-08-08:
```
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
[aruco_mapping-1] process has died [pid 70288, exit code -6, cmd /home/mic-711/catkin_ws/devel/lib/aruco_mapping/aruco_mapping /image_raw:=/frontNear/left/image_raw __name:=aruco_mapping __log:=/home/mic-711/.ros/log/63dc51d6-34d7-11ee-adad-024293f2edd5/aruco_mapping-1.log].
log file: /home/mic-711/.ros/log/63dc51d6-34d7-11ee-adad-024293f2edd5/aruco_mapping-1*.log

```

I think it's an issue with my remap however am not so how what topics is required. Am using a stereo zed2i camera, running rostopic list gives me several topics, example of the format would be: 

`/zed2i/zed_node/left_raw/image_raw_color`

I tried changing the topics but the same error message still appeared.

this is my launch file:

    
```

    <?xml version="1.0"?>
    
    <launch> 
      <!-- <param name="/use_sim_time" value="true"/> -->
      <!-- RVIZ -->
      <!-- <node name="rviz" pkg="rviz" type="rviz" args="-d $(find aruco_mapping)/launch/aruco_config.rviz" /> -->
    
       <!--   usb_cam node -->
     <!--  <node name="usb_cam" pkg="usb_cam" type="usb_cam_node" output="screen">
        <param name="video_device" value="/dev/video0" />
        <param name="image_width" value="640" />
        <param name="image_height" value="480" />
        <param name="pixel_format" value="mjpeg" />
        <param name="camera_frame_id" value="usb_cam" />
        <param name="io_method" value="mmap"/>
      </node> -->
      
     <!-- <node name="usb_cam" pkg="usb_cam" type="usb_cam_stereo_node" output="screen" >
        <param name="video_device" value="/dev/video1" />
        <param name="image_width" value="2560" />
        <param name="image_height" value="720" />
        <param name="pixel_format" value="yuyv" />
        <param name="camera_frame_id" value="Stereo" />
        <param name="left_camera_name" value="left" />
        <param name="reft_camera_name" value="right" />
        <param name="io_method" value="userptr"/>
        <param name="framerate" value="30" />
     </node>  -->
      <!-- ArUco mapping -->
      <node pkg="aruco_mapping" type="aruco_mapping" name="aruco_mapping" output="screen">
        <remap from="/image_raw" to="/zed2i/zed_node/stereo_raw/image_raw_color"/>
    
        <param name="calibration_file" type="string" value="$(find aruco_mapping)/data/camera_config.ini" /> 
        <param name="num_of_markers" type="int" value="2" />
        <param name="marker_size" type="double" value="0.205"/>
        <param name="space_type" type="string" value="plane" />
        <param name="roi_allowed" type="bool" value="false" />
        <!--
        <param name="roi_x" type="int" value="0" /> 
        <param name="roi_y" type="int" value="0" />
        <param name="roi_width" type="int" value="640" /> 
        <param name="roi_height" type="int" value="480" /> 
      -->
    
      </node>  
      
    
      
    </launch>
```
    
The camera I am using is a zed2i camera connecting to a jetson orin nx, it plugs into the 3.0usb of the orin and c-type of camera. 

Any help would be appreciated thank you.

Superyanzhuang, 2023-08-10:
I also have the same problem, If you have a solution, please let me know. Thank you very much

zat1999, 2023-08-10:
@Superyanzhuang I think it was an error with the parser. It wasn't able to read the value from the distortion matrix properly. As a temporary measure I just hard coded the values from my calibration file distortion value into the main.cpp file.

```
 for(size_t i = 0; i < 5; i++)
    distortion_coeff->at<double>(i,0) = camera_calibration_data.D.at(i);
    //Just comment this out and hard code the values for each d coefficient
```

Also not sure if you have this issue but after remapping the publishing topic from my camera to '/image_raw' in the launch file specified by the subscriber, the new topic didn't appear. To fix that issue I just edit this line in the main.cpp:

```
image_transport::Subscriber img_sub = it.subscribe("/image_raw", 1, &aruco_mapping::ArucoMapping::imageCallback, &obj);  
//to
image_transport::Subscriber img_sub = it.subscribe("/zed2i/node/...", 1, &aruco_mapping::ArucoMapping::imageCallback, &obj);
```

While it's not really a permanent solution, hopefully with this temporary fix you can find a proper solution.

zat1999, 2023-08-10:
also not sure if I should make a new thread for this but it's still a related problem regarding the aruco_mapping launcher. When I try to launch the file, I get an error prior to the line:

`Detector.detect(input_image,temp_markers,aruco_calib_params_,marker_size_);`
in the aruco_mapping.cpp file, line 207

Error message in question:
`[aruco_mapping-1] process has died [pid 70288, exit code -11, cmd /home/mic-711/catkin_ws/devel/lib/aruco_mapping/aruco_mapping`

Any idea what this exit code means?

Superyanzhuang, 2023-08-12:
Thank you for your reply，refer to # 17 and your suggestions, but the issue has not been resolved. Perhaps I need to calm down and take a good look at the code. If you have any other suggestions, please let me know. Thank you

Superyanzhuang, 2023-08-12:
> @Superyanzhuang I think it was an error with the parser. It wasn't able to read the value from the distortion matrix properly. As a temporary measure I just hard coded the values from my calibration [...]
Yes, the parseCalibration() function is not working（aruco_mapping.cpp）, which may be the reason for the error